jac99/MinkLocMultimodal #1: RobotCar Seasons

qiaozhijian, 2021-05-10:
Hello~ Thanks for your work!
How can I reproduce the result of Table 3 in your paper? Can you provide the script that finds LiDAR readings with corresponding timestamps in the in original RobotCar dataset for each image in RobotCar Seasons Datasets?

jac99, 2021-05-10:
Hi,
The evaluation on RobotCar Seasons took multiple steps using a combination of Python and Matlab scripts. I've just published some of these scripts in robotcar_seasons_benchmark subfolder in the repo.

First, for each image in the RobotCar Seasons dataset we build a corresponding point cloud, by finding matching LiDAR scans in the full Oxford RobotCar dataset based on timestaps. We build point clouds using the same procedure as in PointNetVlad paper (merging scans from 20 meter traversal, removing ground plane and downsampling to 4096 points).
You can download this dataset [here](https://drive.google.com/drive/folders/1Z7CpG5_Ylq4NeHHN-otUIA2NcKZDZYso?usp=sharing). Each point cloud is named with a timestamp - the same timestamp as a corresponding RobotCar Seasons image.

To reproduce the results follow these steps:

1. Download the dataset with point clouds corresponding to RobotCar Seasons images from the link above.
2. Run `image_pointcloud_path_mapping.py` (in robotcar_seasons_benchmark folder in this repo) which produces a pickle with the list of triplets: (Robotcar seasons image timestamp, relative path to the image, relative path to the point cloud_path). These will be used later when computing global descriptors.
3. Run the code to read the pickle produced in step 2, and compute global descriptor for each element: load the point cloud and the corresponding image, push it through the trained model and save the embedding. The results should be pickled as a dictionary `embeddings_d[timestamp] = embedding` (256 dimensional ndarray of floats32). Unfortunately I cannot find our code to to this step. But in the link I've given above you can find a pickle with global descriptors computed using our trained multimodal model:` season_scan_embeddings_model_MinkLocMultimodal_20210205_1721.pickle`. You can check what format it should have.
4. Finally, run the script: `estimate_season_poses.py` to estimate poses of query elements in the RobotCar Sesons and export them in the format expected by the RobotCar Seasons submission website. `season_scan_embeddings_model_MinkLocMultimodal_20210205_1721.txt` contain output for our global descriptors.

Some of these macros will print a working that round 40 cloud are missing - but that's expected, as we were not able resonstruct point clouds for about 40 images in RobotCar Seasons dataset.

qiaozhijian, 2021-05-10:
Got it. Thanks!

closed 2021-05-10 by qiaozhijian