jac99/MinkLocMultimodal #2: RGB generation

AirOllie, 2021-05-21:
Thanks for the work!
I'm a beginner in point cloud. When I run generate_rgb_for_lidar.py, it seems that lidar2image_ndx.pickle is required, but I cannot figure out where it is first generated. The problem is reported before the function create_lidar2img_ndx has been run. Could U plz give me some hint about that?

jac99, 2021-05-26:
Hi, `lidar2image_ndx.pickle` is created on the first run of `generate_rgb_for_lidar.py` script. The code checks if the pickle exists and if not it will be created. This pickle contains mapping between point clouds (they can be downloaded using the link given in the readme file) and images in Oxford RobotCar dataset (you need to download RobotCar dataset by yourself). The mapping is done based on timestamps (for each point cloud we find images in Oxford RobotCar dataset with closest time stamps).
Then, in the second step, the script extracts 20 images with the closests timestamp from original Oxford RobotCar dataset, downsamples them and saves in the output directory.

In order to run the script you need to download an original Oxford RobotCar dataset (images from stereo/centre camera are enough). Instructions on RobotCar dataset download are here: https://robotcar-dataset.robots.ox.ac.uk/
Then, you need to `run generate_rgb_for_lidar.py` script. 
`--config` parameter should point to `config/config_baseline_multimodal.txt` file - you need to change paths to `dataset_folder` to point to the folder where point clouds are located.
`--oxford_root` to point to Oxford RobotCar dataset root folder
`--out_path` to point to the folder where extracted and downsampled images corresponding to point clouds will be saved
For other parameters use default values.

jac99, 2021-05-26:
I can provide `lidar2image_ndx.pickle` if you want, but without the Oxford RobotCar dataset it'll be useless.

closed 2021-06-04 by jac99