jac99/MinkLocMultimodal #3: KITTI dataset

1207koo, 2021-07-01:
According to your paper, the database and query split of the KITTI dataset were as follows.

`We take Sequence 00 which visits the same places repeatedly and construct the reference database using the data gathered during the first 170 seconds. The rest is used as localization queries.`

However, most queries did not have a close database data with the above split when I tried. (sequence 00 -> database: first 170 sec & query: rest(=170~470 sec)) Could you please explain more about this setting or send me the code?

wysoczanska, 2021-07-03:
Yes, you can check the whole traversal in KITTI 00 sequence. Technically, there are certain parts being revisited during the sequence, and only the scans which indeed have a close database are taken into account in recall calculation. I pushed the notebook for generating queries on KITTI 00 sequence. 

Red: database part
[image]

1207koo, 2021-07-04:
Thank you for the reply.
This has been a sufficient explanation, and I will close the issue.

closed 2021-07-04 by 1207koo