jac99/MinkLocMultimodal #4: KITTI dataset details

1207koo, 2021-07-27:
I have a few questions regarding the KITTI dataset.

1. There are a total of 4 images (grayscale 2, color 2) in the KITTI dataset. I wonder which one was used.
2. When using the point cloud, did you remove the road and downsampled it to have 4096 points like the oxford and inhouse datasets? Moreover, if so, KITTI's point cloud seems to contain a wider area of ​​information. Did you crop it?
3. How did you handle the northing and easting information? The distance was calculated only by northing and easting in the existing oxford and inhouse datasets without considering the altitude. Did you save and use it in the same way in KITTI? (for example, easting: x, northing: z) Or did you use the 3D distance?

I would be waiting for your reply. Thank you!

wysoczanska, 2021-07-27:
Hi, 
1. We used color images only from Camera2, which corresponds to 'image_2' folder in the original dataset. The Camera2 was mounted in the center of the vehicle, which turns out to be sufficient for this task. 
2. The results presented in the paper are the ones with no prior preprocessing since we want our method to operate on raw point clouds. Moreover, we did not observe any significant difference when removing the ground plane. However, I suspect cropping a cloud could slightly improve the results. 
3. We used only x,z to calculate distances (2D).

1207koo, 2021-07-27:
Thank you for the comment. That was very helpful!

1207koo, 2021-07-27:
I have one more question. I'm trying to run it with the KITTI dataset, but if I put it as is, the number of points does not decrease with ME.utils.sparse_quantize. Is this correct? Or is it correct to shift and rescale to zero mean and [-1, 1] like the existing dataset?

wysoczanska, 2021-07-27:
No, there is no need for that. You can simply perform quantization without rescaling. For Kitti dataset we set quantization step=0.25.

1207koo, 2021-07-27:
Oh, I understand. Thank you again!

closed 2021-07-27 by 1207koo