jac99/MinkLocMultimodal #5: MinkLoc++ (RGB-only) generation

sunmilelj, 2021-09-23:
Hello~ Thanks for your work!
       I saw that your paper gave the result of using only rgb pictures. But I did not find the relevant weight file and runing steps. Can you tell me the operation steps?

jac99, 2021-09-23:
Hi, unfortunately weights for RGB-only model are lost. But I've updated the code in the repo and added support for RGB-only model.

jac99, 2021-09-23:
Hi, I've updated the code in the repo and added RGB-only model. But I found out that there's some problem with the training code, as models are not learning (not only RGB-only model, but also multimodal and unimodal point cloud-based models). whereas evaluation of pre-trained models works OK and gives results as in the paper. I'll look into it shortly.

sunmilelj, 2021-09-23:
Thank you for your reply and look forward to your training code.

jac99, 2021-09-23:
The training code should work OK now. Unfortunately weights for RGB-only model which we used to report results for Kitti (Table II in the paper, results for MinkLoc++ using only RGB) are lost. Also we haven't released evaluation code for Kitti.
But I've added RGB-only model to the repo and the code to train it on RobotCar dataset. So you may train RGB-only model for yourself. The first step is to download and pre-process datasets, as per instructions in the README. To run the training:  
`python train.py --config ../config/config_baseline_rgb.txt --model_config ../models/minklocrgb.txt `
You'll need to change paths in `config_baseline_rgb.txt` file pointing to the dataset.

I've also added weights of the RGB-only model trained on RobotCar datasets to the `weights` folder. However, this is a recently trained model, not the one used to report results in the paper, so the results may be different. You may test this model by running: 
`python evaluate.py --config ../config/config_baseline_rgb.txt --model_config ../models/minklocrgb.txt --weights ../weights/model_MinkLocRGB_20210923_2050_final.pth` . It should report avg. top 1% recall = 96.35.
Note, that this model is trained on RobotCar images (stereo/centre camera) downsampled to 320x240 from original 1280x960 resolution. So the model performance on images with different resolution may be suboptimal.

sunmilelj, 2021-09-24:
> The training code should work OK now. Unfortunately weights for RGB-only model which we used to report results for Kitti (Table II in the paper, results for MinkLoc++ using only RGB) are lost. Also [...]

sunmilelj, 2021-09-24:
Thank you very much, I got it

closed 2021-09-24 by sunmilelj