jac99/MinkLocMultimodal #6: bin file data format

sunmilelj, 2021-09-24:
Hello, I have another question. When I read the point cloud bin file, I found that the value inside is very small, all less than 1. What preprocessing has been done? I would be waiting for your reply. Thank you!

jac99, 2021-09-24:
Hi, points clouds in robotcar dataset, are actually not the original point clouds in Oxford RobotCar dataset. The dataset we use contains pre-processed point clouds from an original Oxfrod RobotCar dataset, prepared by authors of PointNetVLAD paper. Their pre-processed dataset is used in a number of other works (e.g. PCAN, LPD-Net, MinkLoc3D and our MinkLocMultimodal).
This paper describes how these preprocessed point clouds are prepapred: https://arxiv.org/pdf/1804.03492.pdf. Basically they merge consecutive scans from 2D Lidar (from the original Oxford RobotCar dataset) during 20 meter traversal of the car, remove points on the ground plane, downsample to exactly 4096 points **and re-scale all coordinates to be in -1..1 range**. So it's OK that the values are within -1..1 range - it corresponds to a real range about 20 meters.
`mink_quantization_size` parameter in model_config file (e.g. in models/minklocmultimodal.txt) controls voxel size and must be set appropriately. Currently it's set to: `mink_quantization_size = 0.01` - which is fine for point clouds with coordinates in -1..1 range. In each direction we'll have up to 200 voxels.
But if you want to process point clouds with coordinates in real world units (e.g. point clouds from 3D Lidar scans with coordinates within 100 meters range from Kitti dataset) you need to change `mink_quantization_size` parameter to larger value, e.g. 0.3 (=30 centimeters).

sunmilelj, 2021-09-24:
Ok, I see, it’s great for me to help the team, thank you very much

closed 2021-09-24 by sunmilelj