jac99/MinkLocMultimodal #9: Question about the code

LZL-CS, 2021-11-21:
Hi, I am confused about the `positives_masks` and `negative_masks` could you please explain more about it?
https://github.com/jac99/MinkLocMultimodal/blob/683ef1aae35ab1b60f13cefccfdd0e3f9cb9ea6e/datasets/dataset_utils.py#L48

jac99, 2021-11-22:
Hi,

training loop process batches of training elements (point clouds) of size N.
This fragment of code, in `collate_fn`, for each batch of N point clouds, constructs NxN boolean masks of positive and negative examples within a batch.
`positives_mask[i, j]` is True if _j_-th batch element is a positive example for _i_-th batch element (that is, if the ground truth distance between the center of _i_-th and _j_-th point clouds is below a 10-meter threshold).
`negatives_mask[i, j]` is True if _j_-th batch element is a negative example for _i_-th batch element (that is, if the ground truth distance between the center of _i_-th and _j_-th point clouds are below a 50-meter threshold).
See section _B. Network training_ in the paper, for description of positive and negative examples.
positives_mask and negatives_mask arrays are constructed, based on information generated using `generate_training_tuples_baseline.py` script. This script for each point cloud finds a set of positive and non-negative examples and saves them in a pickle.

`BatchSampler` (in `samplers.py`) constructs training batches, by sampling pairs of positive elements from the training set. So if you inspect 'positives_mask' array, elements [0, 1] and [1,0] should be True (as the first and second elements are a positive pair). Same for elements [2,3] and [3,2]. Other elements should be mostly False, but due to the random sampling in BatchSampler, it may happen. than for some i-th batch element there're more positives, so there'll be more True values in `positives_mask` array.

LZL-CS, 2021-11-23:
> Hi,
> 
> training loop processes batches of training elements (point clouds) of size N. This fragment of code, in `collate_fn`, for each batch of N point clouds, constructs NxN boolean masks of positive and negative examples within a batch. `positives_mask[i, j]` is True if _j_-th batch element is a positive e for _i_-th batch element (that is, if the ground truth distance between the center of _i_-th and _j_-th point clouds is below a 10-meter threshold). `negatives_mask[i, j]` is True if _j_-th batch element is a negative for _i_-th batch element (that is, if the ground truth distance between the center of _i_-th and _j_-th point clouds are above a 50-meter threshold). See section _B. Network training_ in the paper, for a description of positive and negative examples. `positives_mask` and `negatives_mask` arrays are constructed, based on information generated using `generate_training_tuples_baseline.py` script. This script finds a set of positive and non-negative examples for each point cloud and saves them in a pickle.
> 
> `BatchSampler` (in `samplers.py`) constructs training batches, by sampling pairs of positive elements from the training set. So if you inspect `positives_mask` array, elements [0, 1] and [1,0] should be True (as the first and second elements are a positive pair). Same for elements [2,3] and [3,2]. Other elements should be mostly False. But due to the random sampling in BatchSampler, it may happen that for some _i_-th batch element, there're more positives, so there'll be more True values in `positives_mask` array.
> 
> These `positives_mask` and `negatives_mask` arrays are used by a triplet loss function. We construct training triplets, by choosing one hardest positive and negative example for each batch element (anchor in a triplet loss function). These boolean masks tell us, from which elements we can choose for each anchor.

ok, I got it. Very thanks for your reply!

LZL-CS, 2021-11-25:
> Hi,
>
> training loop processes batches of training elements (point clouds) of size N. This fragment of code, in `collate_fn`, for each batch of N point clouds, constructs NxN boolean masks of positive [...]

ok, I got it. Very thanks for your reply!

closed 2021-11-25 by LZL-CS