jac99/MinkLocMultimodal #11: loss become nan during training

FANzhaoxin666, 2021-11-28:
hello!
 I run your code and find the loss becomes nan after 40 epoch. I find it is caused by that values of all embeddings become nan.  Do you find this during your research? Is there any bug? or just by accident?

jac99, 2021-11-28:
Hi,
I haven't encountered this in my experiments. Can you post details of _config_ and _model_config_ files, used for the training? Can you also post a screenshot of the output from the training process, just before a loss becomes nan (e.g. screenshot showing output for epochs 35 till 42 or similar).

FANzhaoxin666, 2021-11-28:
> Hi, I haven't encountered this in my experiments. Can you post details of _config_ and _model_config_ files, used for the training? Can you also post a screenshot of the output from the training [...]

Thank you. Sorry for that I didn't save the log. The config and model_config are the same with you. We retrained the model for several times. We find it is the image_loss become nan. And the total loss kept to be 0.2. We also find that there is no batch zise expanding during training. We suspect it may be caused by pytorch version. We use pytorch 1.8 but we notice you use 1.9.1. Before using pytorch 1.8, we also tried to use pytorch 1.7, but there would be an error: the positive/negative mask value must be bool type. Therefore, we think   it is the different pytorch version try the positive/negative mask differently caused the problem.  We are now trying to replace the pytorch version to 1.9 or 1.10. It there any new findings, we will tell you. Thanks.

FANzhaoxin666, 2021-11-28:
[image]
Here is my config.

FANzhaoxin666, 2021-11-28:
[image]
He is a screen shot of the newest training we tried.

jac99, 2021-11-28:
If the loss is close to the margin (0.2) and the number of active triplets is equal to the initial batch size and does not decrease (initial batch size = 8 and the number of active triplets is 8) this indicates some problem in the training process. Likely all embeddings collapse to zero and the network is not training. After a few epochs, all embeddings become zero and this causes NaNs.
The expected behaviour is that the number of active triplets starts decreasing from epoch 1. Then, when it decreases below a threshold (when there's less than 70% of active triplets), the batch size is automatically increased (until it reaches batch_size_limit=160). 
I suggest to try training the unimodal model first (point cloud only model): 
`python train.py --config ../config/config_baseline.txt --model_config ../models/minkloc3d.txt`
If it trains correctly, and evaluation at the end of the training gives expected results, it means there's likely some issue with training images.
I'll try to clone the github repo later and run the training again, to verify that the github code works correctly.

jac99, 2021-11-28:
Also ensure that you are using the latest version of the code. Some time ago there were some changes done in the repo, and older versions (like 3 months old or so) may not work correctly.

FANzhaoxin666, 2021-11-28:
Thank you! The problem is solved by replacing the code you released 2 months ago with the latest version. Thank you so much.

jac99, 2021-11-29:
Issue is solved.

jac99, 2021-11-29:
Issue is solved, closing.

closed 2021-11-29 by jac99