jac99/MinkLocMultimodal #12: Question about model structure

LZL-CS, 2022-02-23:
Hi,
I recently implemented your work, but I am confused about your model structure, which is shown as blew:
[image]

Question-1: I think the module of **convs(1)** is not consistent with your description in Fig.3 of your paper?
Question-2: What does the module **downsample** for?

jac99, 2022-02-24:
Hi,

regarding Question 1. Yes, you are right. The network architecture in the code does not correspond to Table 4 in the paper. 
The layer in red is the first layer in Conv2 block. In Table 4 it is described as C with upper index 64 (meaning 64 convolutional filters = output channels) and lower index 2k2s = 2x2x2 kernel with 2x2x2 stride.
Whereas, in the highlighted code, it has only 32 output channels, not 64.
I also include a mapping between network architecture in the code and Table 4.

[image]

jac99, 2022-02-24:
Regarding question 2: downsample module is used in the residual connection. See code of the class `ECABasicBlock(BasicBlock)`. The input to the residual block in block Conv2 has 32 channels (not 64 as you've noticed). And the output (after passing through 2 convolutional layers and ECA attention layer) has 64 channels. To make a residual connection we need to transform this 32 channel input to 64 channels - to be able to add it to the output of ECA attention layer.
Actually the name is misleading, we are not doing any downsampling, we actually increase the number of channels.

closed 2022-02-24 by jac99