jac99/MinkLocMultimodal #13: How to normalize the RobotCar pointcloud data to [-1,1]?

LZL-CS, 2022-04-15:
Hi, in your paper **"As point coordinates in the Oxford RobotCar dataset are normalized to be within [−1,1] range, this gives up to 200 voxels in each spatial direction."** May I know what algorithm was used to get the normalized point cloud between -1 and 1?

jac99, 2022-04-15:
Hi, we used Oxford RobotCar point clouds prepared by the authors of PointNetVLAD paper. Here also provided Matlab script generating the data. See: https://github.com/mikacuy/pointnetvlad/blob/master/submap_generation/generate_submaps.m

In lines 189-210 they do this point cloud normalization. Basically, they compute the point cloud centroid and shift all points so that its center has (0,0,0) coordinates. And multiply by some scaling factor so x,y coordinates fell into -1..1 range.

LZL-CS, 2022-04-16:
ok, I will try it, thank you very much for your reply.

LZL-CS, 2022-04-16:
Hi, in your work you used the 2D LiDAR (from Oxford RobotCar: 2 x SICK LMS-151 2D LIDAR, 270° FoV, 50Hz, 50m range, 0.5° resolution). According to the documentation description: **Each 2D scan consists of 541 triplets of (x, y, R), where x, y are the 2D Cartesian coordinates of the LiDAR return relative to the sensor (in metres), and R is the measured infrared reflectance value.**  Does the point cloud data you used actually not a real 3-dimensional dataset, the coordinate **Z** represents the reflectance.

closed 2022-04-18 by LZL-CS