jac99/MinkLocMultimodal #14: How to evaluate on KITTI dataset?

Kartikgar, 2022-04-17:
Hi, I wanted to recreate the generalisation results on KITTI dataset, that you have mentioned in the paper. I would appreciate any advice on how to run the model for KITTI dataset.

jac99, 2022-04-18:
Hi,
we haven't released the code for evaluation on Kitti and we have no plans to release it.
But you can find some details regarding evaluation in Kitti in these posts, maybe they'll be useful:
https://github.com/jac99/MinkLocMultimodal/issues/3
https://github.com/jac99/MinkLocMultimodal/issues/4

For Kitti dataset, we use point clouds built from a single scan of 3D LiDAR with 360 degrees field of view.
You can find a dataloader for Kitti point clouds (but not for images) in our later work EgoNN. See here: https://github.com/jac99/Egonn/tree/main/datasets/kitti

Kartikgar, 2022-04-29:
Hi,
Thank you for pointing me to Egonn, it helped me out. I just wanted to confirm if MinkLoc++ uses the Kitti dataset in the same way Egonn does in terms of making the map and queries.

closed 2022-05-17 by jac99