jac99/MinkLocMultimodal #15: Question about pointcloud data

LZL-CS, 2022-04-18:
Hi, in your work you used the 2D LiDAR (from Oxford RobotCar: 2 x SICK LMS-151 2D LIDAR, 270° FoV, 50Hz, 50m range, 0.5° resolution). According to the documentation description: Each 2D scan consists of 541 triplets of (x, y, R), where x, y are the 2D Cartesian coordinates of the LiDAR return relative to the sensor (in metres), and R is the measured infrared reflectance value. Does the point cloud data you used actually not a real 3-dimensional dataset, the coordinate Z represents the reflectance?

jac99, 2022-04-18:
Hi, in our work (MinkLocMultimodal) we do not use raw scans from 2D LiDAR. We use the same point clouds as introduced in PointNetVLAD paper (see section 5.1 in their paper: https://arxiv.org/pdf/1804.03492.pdf). These point clouds are constructed by **accumulating multiple 2D LiDAR scans as the vehicle moves for a 20-meter distance**. For each scan, we take only (x,y) coordinates and ignore the reflectance value. Our paper and PointNetVLAD paper show visualization of these point clouds - you can see parts of the buildings and other structures.
In contrast, our later work EgoNN (https://github.com/jac99/Egonn), uses point clouds built from a single scan of 3D LiDAR.

jac99, 2022-04-18:
Nota bene, the same points clouds (accumulated 2D LiDAR scans, as the vehicle moves for 20 meters; downsampled to 4096 points and normalized in -1..1 range; with the ground plane removed) are used in a number of other papers. See tables at the bottom of this page: https://github.com/jac99/MinkLoc3Dv2 - all these methods use the same point clouds.

LZL-CS, 2022-04-21:
Hi@jac99, thank you very much for your reply, I will read your other papers for more information. But I am still confused about some issues:
1. I know the file (e.g., "pointcloud_locations_20m_10overlap.csv" stores the XY coordinate locations of pointcloud), but the folder(e.g., "pointcloud_20m_10overlap" stores the XYZ coordinate of pointcloud ), so why 2D-LiDAR can obtain XYZ coordinate without using reflectance? 
[image]
[image]

2. In your code: https://github.com/jac99/MinkLocMultimodal/blob/e8fc6d23c3c6c3cb937186fd0efeb2b67a67b636/training/trainer.py#L107 you take `'lr': params.image_lr` into `params_l` for model training. I can't figure out why you do this? In my naive opinion, the learning rate has been tuned by code: https://github.com/jac99/MinkLocMultimodal/blob/e8fc6d23c3c6c3cb937186fd0efeb2b67a67b636/training/trainer.py#L137
Looking forward to your reply, thanks in advance.

LZL-CS, 2022-04-21:
ok, thanks for your reply

closed 2022-04-21 by LZL-CS