jac99/MinkLocMultimodal #19: the Oxford Robotcar Dataset unavailable

where2go947, 2022-11-27:
Hi, thanks for your great work on multi modal fusion for place recognition. 
However, the Oxford Robotcar Dataset is [unavailable from 2022-11-08](https://robotcar-dataset.robots.ox.ac.uk/) and I'm a bit frustrated that there is no sign of re-opening the download. Could you please provide me with the centre RGB images by Google drive?
Thanks in advance.

jac99, 2022-11-27:
Hi, I've uploaded pre-processed images from RobotCar dataset that are required to train and evaluate MinkLocMultimodal. In the below Google Drive link, you'll find compressed content of the folder containing downsampled RobotCar images from centre camera. 
After downloading and unpacking, set `image_path` parameter in `config_baseline_multimodal.txt` to point to the folder with these images.
https://drive.google.com/file/d/1fnpBW57LVLZxOZU5-dSnuwfdwbfNyxGg/view?usp=sharing

where2go947, 2022-11-28:
Many thanks for your kind and warm help.

closed 2022-11-28 by where2go947